OctoPrint/OctoPrint #5177: Calling /api/settings breaks reset of plugin settings with {} default

jacopotediosi, 2025-07-31:
### The problem

After the `/api/settings` API is called, a plugin setting that has `{}` as its default can no longer be reset to that default value.

### Steps to reproduce:

Install the following plugin: 
```python
import octoprint.plugin

class settingstestPlugin(octoprint.plugin.SettingsPlugin, octoprint.plugin.BlueprintPlugin):
    def get_settings_defaults(self):
        return dict(
            test={},
        )
        
    @octoprint.plugin.BlueprintPlugin.route("/get", methods=["GET"])
    def myGet(self):
        return str(self._settings.get(["test"]))
    
    @octoprint.plugin.BlueprintPlugin.route("/set", methods=["GET"])
    def mySet(self):
        self._settings.set(["test"], {"A":"B"})
        self._settings.save()
        return "DONE"
    
    @octoprint.plugin.BlueprintPlugin.route("/reset", methods=["GET"])
    def myReset(self):
        self._settings.set(["test"], {})
        self._settings.save()
        return "DONE"

__plugin_name__ = "settingstest"
__plugin_pythoncompat__ = ">=3.7,<4"
__plugin_implementation__ = settingstestPlugin()
```

Here’s a pre-packaged zip to easily install it via the Plugin Manager:  
[settingstest.zip](https://github.com/user-attachments/files/21529587/settingstest.zip)

Make sure to restart OctoPrint but do not open the web UI, to avoid triggering `/api/settings` API calls.

Below is a quick usage of the plugin’s API to confirm everything behaves as expected:
```bash
> export APIKEY=<your api key here>

> curl -H "Authorization: Bearer $APIKEY" http://127.0.0.1:5000/plugin/settingstest/set
DONE

> curl -H "Authorization: Bearer $APIKEY" http://127.0.0.1:5000/plugin/settingstest/get
{'A': 'B'}

> curl -H "Authorization: Bearer $APIKEY" http://127.0.0.1:5000/plugin/settingstest/reset
DONE

> curl -H "Authorization: Bearer $APIKEY" http://127.0.0.1:5000/plugin/settingstest/get  
{}
```

Now, repeat the test, but this time call the `/api/settings` API before triggering the `/plugin/settingstest/reset` operation:
```bash
> curl -H "Authorization: Bearer $APIKEY" http://127.0.0.1:5000/plugin/settingstest/set  
DONE

> curl -H "Authorization: Bearer $APIKEY" http://127.0.0.1:5000/plugin/settingstest/get
{'A': 'B'}

> curl -H "Authorization: Bearer $APIKEY" http://127.0.0.1:5000/api/settings
(...response omitted...)

> curl -H "Authorization: Bearer $APIKEY" http://127.0.0.1:5000/plugin/settingstest/reset
DONE

> curl -H "Authorization: Bearer $APIKEY" http://127.0.0.1:5000/plugin/settingstest/get
{'A': 'B'}
```

As shown in the example above, after calling the `/api/settings` API, the `test` setting can no longer be reset to `{}` — it stays stuck at its previous value. It can still be changed to any other value, just not back to its default.  
I verified that this issue only occurs when the setting's default value is `{}`.

It’s worth noting that `{}` is correctly saved in the `config.yaml` file, so the problem lies only in the value returned at runtime by `self._settings.get()`, which no longer reflects the saved settings.

### Proposed solution

I don't know yet.  
However, I assume this is a bug, since performing a GET request to `/api/settings` shouldn't alter the settings structure or cause any side effects.

### Additional information

I was able to reproduce this bug on Linux, both in the latest commit of the `maintenance` branch and in release [1.11.2](https://github.com/OctoPrint/OctoPrint/releases/tag/1.11.2).
To ensure the issue wasn't caused by other plugins or custom configurations, I also reproduced it in a fresh [docker container](https://github.com/OctoPrint/octoprint-docker).

I didn't get any relevant logs, even after setting OctoPrint's logging level to DEBUG.

### Verification

- [x] I'm a maintainer or collaborator of OctoPrint and allowed to use this issue type

jacopotediosi, 2025-07-31:
After some testing, I found that the bug occurs when `settings().effective` is called by the `/api/settings` decorator at this point:  
https://github.com/OctoPrint/OctoPrint/blob/89733fd12fbd1b4ad9be08ee1f4ccafefdafe340/src/octoprint/server/api/settings.py#L85

jacopotediosi, 2025-08-02:
The issue appears to be that the current implementation of [`_unflatten()`](https://github.com/OctoPrint/OctoPrint/blob/89733fd12fbd1b4ad9be08ee1f4ccafefdafe340/src/octoprint/settings/__init__.py#L205) modifies the input dictionary.

I confirmed this by adding logging lines to the function:
```python
    def _unflatten(d: dict[str, Any], prefix: str = "") -> dict[str, Any]:
        """
        Unflattens a flattened dictionary.

        Args:
            d (dict): The flattened dictionary.
            prefix (str): The prefix to use for the top level keys. If provided, only keys
                starting with this prefix will be unflattened and part of the result.

        Returns:
            dict: The unflattened dictionary.
        """
        
        original_d = fast_deepcopy(d)

        if d is None:
            return {}

        result = {}
        for key, value in d.items():
            if not key.startswith(prefix):
                continue
            subkeys = key[len(prefix) :].split(_CHAINMAP_SEP)
            current = result

            path = []
            for subkey in subkeys[:-1]:
                # we only need that for logging in case of data weirdness below
                path.append(subkey)

                # make sure the subkey is in the current dict, and that it is a dict
                if subkey not in current:
                    current[subkey] = {}
                elif not isinstance(current[subkey], dict):
                    logging.getLogger(__name__).warning(
                        f"There is a non-dict value on the path to {key} at {path!r}, ignoring."
                    )
                    current[subkey] = {}

                # go down a level
                current = current[subkey]

            current[subkeys[-1]] = value
            
        if d != original_d:
            logging.critical("_unflatten() modified the input dictionary!")
            logging.critical(f"Original: {original_d}")
            logging.critical(f"Modified: {d}")

        return result
```

Output:
```
2025-08-02 17:36:18,086 - root - CRITICAL - _unflatten() modified the input dictionary!
2025-08-02 17:36:18,086 - root - CRITICAL - Original: {'plugins\x1fsettingstest\x1ftest': {}, 'plugins\x1fsettingstest\x1ftest\x1fA': 'B'}
2025-08-02 17:36:18,086 - root - CRITICAL - Modified: {'plugins\x1fsettingstest\x1ftest': {'A': 'B'}, 'plugins\x1fsettingstest\x1ftest\x1fA': 'B'}
```

The reason the bug was triggered by calling the `/api/settings` API is that the API invokes `_unflatten()` through the following call stack:

[getSettings() ](https://github.com/OctoPrint/OctoPrint/blob/89733fd12fbd1b4ad9be08ee1f4ccafefdafe340/src/octoprint/server/api/settings.py#L114) --> decorated by [@with_revalidation_checking](https://github.com/OctoPrint/OctoPrint/blob/89733fd12fbd1b4ad9be08ee1f4ccafefdafe340/src/octoprint/server/api/settings.py#L107) --> calls [_etag()](https://github.com/OctoPrint/OctoPrint/blob/89733fd12fbd1b4ad9be08ee1f4ccafefdafe340/src/octoprint/server/api/settings.py#L108) --> calls [settings().effective](https://github.com/OctoPrint/OctoPrint/blob/89733fd12fbd1b4ad9be08ee1f4ccafefdafe340/src/octoprint/server/api/settings.py#L85) --> calls [deep_dict()](https://github.com/OctoPrint/OctoPrint/blob/89733fd12fbd1b4ad9be08ee1f4ccafefdafe340/src/octoprint/settings/__init__.py#L988C17-L988C18) --> calls [_unflatten()](https://github.com/OctoPrint/OctoPrint/blob/89733fd12fbd1b4ad9be08ee1f4ccafefdafe340/src/octoprint/settings/__init__.py#L321)

It looks like the bug can be fixed by modifying:
https://github.com/OctoPrint/OctoPrint/blob/89733fd12fbd1b4ad9be08ee1f4ccafefdafe340/src/octoprint/settings/__init__.py#L245

To:

```python
current[subkeys[-1]] = fast_deepcopy(value)
```

For performance reasons, of course `fast_deepcopy` could be applied only if `value` is an instance of `dict`.

foosel, 2025-09-01:
Thank you very much for your detective work on that one! Your analysis was spot on! 

I had to think a bit about whether adding a `fast_deepcopy` here would be enough or if I actually also needed to fix that there's an empty dict in the data, but then I realised that that actually comes from the default of the settings, and the version with data in it comes from the current config data. So both existing at the same time is normal and ok, and should be handled correctly without side effects. 

I've added the situation to the existing unit tests of `_unflatten` and also added a check to the test that the original data isn't being changed, and also fixed the issue on `staging/bugfix`, so it will be part of the upcoming 1.11.3 bugfix release.

commit e33a84dd (2025-09-02): fix: remove side effect on HierarchicalChainMap._unflatten

Closes #5177

Huge thanks to @jacopotediosi for the detective work on that!
closed 2025-09-09 by foosel