OctoPrint/OctoPrint #5196: OctoPrint does not recognise the full file length of very large files

nicclift, 2025-09-17:
### The problem

I am printing a large file of a flexible dragon, ~248MB. 

It stops prematurely, always at the same place, at around 55% complete (as per the printer), but shown to be 100% by Obico; and by OctoPrint's own internal progress tracker. 

The GCode visualiser fails to show an image, sticking during the analysis step at around 55%.  

Print Time Genius (when in use) indicates the same % completion as OctoPrint, which is logical since it presumably won't bother re-inventing this when it is available from the API.

This is a "new" problem - the Gcode visualiser used to work (although it was pretty slow to finish the analysis step, and this same file used to print to the end. This was many months ago - I have not printed this dragon recently until now.

### Did the issue persist even in safe mode?

Yes, it did persist

### If you could not test in safe mode, please state why ("currently printing" is NOT an excuse!)

Switched to Safe Mode and restarted OctoPrint while printing direct from USB memory (not controlling with OctoPrint), and it continued to show the same incorrect % completion

### Version of OctoPrint

1.11.3

### Operating system running OctoPrint

Python 3.11.2 OctoPi* 1.1.0 (build 2025.05.06.135300)

### Browser and version of browser, operating system running browser

Safari Version 26.0 (21622.1.22.11.14); macOS Tahoe Version 26.0 (25A354) (but same problem with macOS 15.x

### Printer model & used firmware incl. version

Prusa XL 5T 6.2.5 (8912)

### Checklist of files to include below

- [x] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20250918081218.zip](https://github.com/user-attachments/files/22395155/octoprint-systeminfo-20250918081218.zip)

Unable to upload GCODE file, available on request

nicclift, 2025-09-17:
Unable to upload the GCODE file - shows on macOS as 245.9MB; 82MB zipped; 100.6MB as bgcode; 99.2MB as bgcode zipped

jneilliii, 2025-09-17:
so you did try with standard gcode file and not bgcode format? running in safe mode wouldn't allow the bgcode upload, so I suspect that is what you did for testing in safe mode.

nicclift, 2025-09-17:
Note that the last time this worked, prior to this overestimation of % completion, it was using bgcode in any case.

This current print was started directly from the Prusa controls, running from a USB drive. I thought it was bypassing OctoPrint entirely. However, Obico continued to send me relevant messages and was showing live video. I stopped and reset the system several times and then restarted it in safe mode. The printer continued to operate. When OctoPrint restarted in safe mode, it was still showing (at that stage) 83% complete, with the printer physical screen showing 47%. I eventually used the OctoPrint browser interface to disconnect. The print is now past the point at which the most recent OctoPrint managed tries have simply stopped (with Obico showing "Finished"), and continuing to be managed by the printer's own internal controls

nicclift, 2025-09-17:
The print takes more than 12h to reach this failure point. This makes this testing very time consuming, prevents other activity, and uses consumables. If the problem is a function of file size (as I'm guessing) it should be relatively easy to find it in the code. It may even have been a programming decision, making an estimate of the upper size of expected files

jneilliii, 2025-09-18:
You can test the process using the virtual printer, it doesn't use consumables, but will take time as it mimics the entire print process, although maybe a little quicker than real-world experience.

nicclift, 2025-09-18:
Thanks - will try that when the current print completes. Currently every time I restart the server on my Pi, it grabs the printer, despite having auto connect disabled in both the dashboard browser screen and in the settings. This print is now over 20h old and I will be happy to let it finish in the next 14h or so before investigating further

foosel, 2025-09-18:
> It may even have been a programming decision, making an estimate of the upper size of expected files

For the record, there is no such thing.

Just so I understand correctly what is happening here, the whole print stops at around 55%, not just the gcode visualizer, right? Does OctoPrint show the full size in the file manager?

Tbh, I'm seriously suspecting some weird shenanigans with bgcode here. Having access to the file, ideally both gcode and bgcode, would certainly help, so if you could drop it here: https://cloud.octoprint.org/s/7EkY7eJw9Gd3QjK that would be lovely.

nicclift, 2025-09-18:
> foosel

Hi Gina

Thank you for such very prompt attention! No critique intended re the limited file size! Noted no specific upper limit

Yes, the file manager shows 248MB, similar to the count on MacOS on my desktop at 245.9MB. 

The gcode visualiser struggles to 55% Analysis and does not get past there - no visualiser image emerges. Works fine for everything else I print - all much smaller. 

The whole print stops at layer 41. So far, several times, all at night. I just find it in the morning with Obico saying it’s finished, the printer sitting unresponsive as though finished, no errors or flags. This morning I happened to wake in the early hours and saw that Obico was flagging over 80% complete, which I knew was not possible. Came out to the printer and found it showing >40%, consistent with rough estimate from visual observation of the advancement of the print. OctoPrint on my browser gave the same (exaggerated) figure as Obico. I estimate that Obico/OctoPrint would arrive at 100% at approximately the time that the layer count would reach 41. This is the first indication of numbers, the previous failures were notably without concrete stats other than the layer height

I did this last run using a USB drive, observing with PrusaConnect and had not expected OctoPrint or Obico to be involved at all. Managed to shut down OctoPrint without crashing the printer and resumed with PrusaConnect. It’s still printing, expected complete in around 10h.

I have forwarded bgcode and gcode versions of the file.My last successful print of this dragon was in June. Same hardware and firmware. That was bgcode. The Mac was still on OS15.x both when last successful and when last failed.

I have forwarded the two files bgcode and gcode as you asked. 

Thanks
Nic

foosel, 2025-09-18:
Got the files, thank you!

So, I'm slightly confused here... Looking at the file, layer 41 is around 8mm in height, and *far* away from 55% and ~20mm, even according to the printer. 

The printer should report 55% between lines 5548902 and 5656508 (location of first `M73 P55` vs first `M73 P56`) and with 10607674 lines all in all that also makes sense (~52% of the whole file looking at line count):

```
$ du -h issue_5196.gcode
235M    issue_5196.gcode
$ wc -l issue_5196.gcode
10607674
$ grep -n "^M73 P55" issue_5196.gcode | head -n 1
5548902:M73 P55 R999
$ grep -n "^M73 P56" issue_5196.gcode | head -n 1
5656508:M73 P56 R977
```

The actual height at this point should be around 20mm (give or take a bit):

```
$ sed -n '5548902,5656508p;5656509q' issue_5196.gcode > issue_5196.partial.gcode
$ grep Z issue_5196.partial.gcode | head -n 10
G1 X319.042 Y204.144 Z20.17 F24000
G1 X321.71 Y202.003 Z20.2
G1 Z19.6 F720
G1 X329.611 Y210.648 Z19.781 F24000
G1 Z19.6 F720
G1 X308.736 Y232.99 Z20.17 F24000
G1 X305.358 Y233.529 Z20.2
G1 Z19.6 F720
G1 X238.59 Y241.938 Z19.674 F24000
G1 Z19.6 F720
```

The file has 492 layer changes all in all and layer 41 corresponds to line 2191995 and Z8.2:

```
$ grep -n ";LAYER_CHANGE" issue_5196.gcode
     1  3756:;LAYER_CHANGE
     2  57670:;LAYER_CHANGE
     3  130364:;LAYER_CHANGE
     4  202484:;LAYER_CHANGE
     5  244699:;LAYER_CHANGE
     6  285526:;LAYER_CHANGE
     7  326550:;LAYER_CHANGE
[...]
    37  1917233:;LAYER_CHANGE
    38  1986169:;LAYER_CHANGE
    39  2055126:;LAYER_CHANGE
    40  2124324:;LAYER_CHANGE
    41  2191995:;LAYER_CHANGE
    42  2260308:;LAYER_CHANGE
    43  2327530:;LAYER_CHANGE
    44  2393001:;LAYER_CHANGE
    45  2458066:;LAYER_CHANGE
    46  2524210:;LAYER_CHANGE
[...]
    96  5439756:;LAYER_CHANGE
    97  5483312:;LAYER_CHANGE
    98  5527309:;LAYER_CHANGE
    99  5571033:;LAYER_CHANGE
   100  5614656:;LAYER_CHANGE
   101  5657153:;LAYER_CHANGE
   102  5699645:;LAYER_CHANGE
[...]
   486  10605637:;LAYER_CHANGE
   487  10606118:;LAYER_CHANGE
   488  10606384:;LAYER_CHANGE
   489  10606665:;LAYER_CHANGE
   490  10606880:;LAYER_CHANGE
   491  10607078:;LAYER_CHANGE
   492  10607198:;LAYER_CHANGE
$ cat issue_5196.gcode | head -n 2192000 | tail -n 10
G1 X10.382 Y314.914 E.18013
G1 X10.828 Y315.044 E.01572
G1 X13.885 Y311.986 E.14636
M204 P2500
;LAYER_CHANGE
;Z:8.2
;HEIGHT:0.2
;BEFORE_LAYER_CHANGE
G92 E0.0
;8.2
```

According to the next `M73` that's at 23% completion (reported on the printer). 55% is around layers 98/99/100.

So something here doesn't add up, your observed layer number doesn't match the percentage reported by the printer (which it takes from the file). Can you please measure your failed prints and tell me the actual height they stopped at? Also I'd like to ask you to disable both PrintTimeGenius and DisplayLayerProgress while we are trying to get to the bottom of this, to rule out any side effects of their progress stuff here. About Obico, I have no clue what is going on there, and I also [am not too keen anymore on supporting them in any way after what they did](https://octoprint.org/blog/2024/07/04/more-stats-manipulation/).

The fact that the analysis in the gcode viewer stops at 55% (which means 10% of the file actually - the status bar runs from 0 to 50% for the parsing and then from 50% to 100% for the analysis) is interesting. I was able to reproduce this here, and am currently waiting for a second try to get to that point, as I now have told the browser to pause the analysis worker if it runs into an exception. My guess is something crashes and it never makes it into the browser log (web workers are a tad unhelpful in that regard). It might just be running out of resources or the browser might decide enough is enough, but it could also be hint, so we'll see.

@jneilliii how did the bgcode plugin work again? It's just an upload preprocessor, right? No runtime changes during printing?

jneilliii, 2025-09-18:
> [@jneilliii](https://github.com/jneilliii) how did the bgcode plugin work again? It's just an upload preprocessor, right? No runtime changes during printing?

Correct, it is just a preprocessor that uses their library to extract the gcode. After upload if you download the file back from OctoPrint you'll see it's actually not binary, just standard text based gcode but with a bgcode extension.

foosel, 2025-09-18:
> I assume no issue running bgcode (presumably your system simply translates this back to GCODE - certainly has done so successfully for every other print I have done - other than the dragon - since you made it available)?

That was the work of @jneilliii, not mine ;) But yes, as just clarified above, the bgcode plugin translates the uploaded bgcode file back into regular gcode on upload, so it would also be good if you could upload the gcode file that you are actually printing: download it from OctoPrint's file manager please, [and share it here](https://cloud.octoprint.org/s/MEA9pWniKk4sZxY) (new sharing link, I already disabled the previous one, goes to the same folder though). Out of curiousity, have you so far *exclusively* printed the bgcode file when this problem arouse, or rather the gcode generated from that? Or *is* the gcode you shared actually the one that's also stored in OctoPrint? I'm sorry to say that I don't know if bgcode will also re-convert comments, hence the question. The difference in file size between your shared gcode and what's being reported to you from the file in question in your OctoPrint instance however makes me think those are not the same files.

> Also disable gcode viewer?

That shouldn't have any impact at all, especially not if you don't even let it load the file.

What would also be interesting is what OctoPrint supports on the frontend when this happens with regards to printed vs total size. So far we are only talking about percentages here, but I'd love to know is whether OctoPrint actually thinks it has read all 234.5MB of data when it reports 100%, or what it says there. Do you remember by any chance?

Side note: this has time, no need to lose sleep over it, good night ;)

foosel, 2025-09-18:
> I was able to reproduce this here, and am currently waiting for a second try to get to that point, as I now have told the browser to pause the analysis worker if it runs into an exception.

Apparently I was not patient enough, the second try is now sitting at 57%, and I can also see when I put a breakpoint in the dev tools it's still running and running and running.

Let's focus on the other points first though, the GCODE viewer is fully client side and shouldn't influence what's printed nor the progress.

nicclift, 2025-09-18:
> foosel
>  left a comment
> (OctoPrint/OctoPrint#5196)
> Well, one good thing has already come out of this. I discovered a very expensive calculation in the analyzer that was actually not used in the UI at all (anymore), so I removed it and now the viewer actually finishes the analysis on this model fast and also fully. So what I can tell you already is, the gcode itself can be visualized again:
>
> https://github.com/user-attachments/assets/7ea40bab-b834-4fe7-8939-f988ca07d1dc
>
> I have no clue why things got worse here, maybe some browser changes? The code in question hasn't been touched for 2 years, last big change was in 1.9.0, and the expensive bit was in there since the start. In any case, that expensive bits are now removed and with 1.11.4 or 1.12.0 (whatever comes first) the gcode viewer will be more performant in that regard.

Hi Gina

I’m pleased you have resolved an issue with the viewer - it’s a nice feature when it works, so this is great news

This morning I woke to a perfect print of the dragon - no defects whatsoever, and no process stops other than a series of brief pauses 24h ago to try to stop any link to OctoPrint in order to test the printer without it. One run is not a statistical sample, but it does indicate that the printer can satisfactorily complete this item.

I have upgraded my Prusa XL to firmware 6.2.6 this morning, from 2.6.5.  It’s been three weeks since its release, and no signs of significant problems. I had already upgraded to PrusaSlicer 2.9.3 before sending you those files.

I’m ready to resume with OctoPrint in order to do the next run:
I will replace the same 3MF project file to produce and transfer GCODE not bgcode, as requested
Disabled PrintTimeGenius; DisplayLayerProgress; and Obico

This next run will be on the actual machine, with filament. I have no experience of testing via your virtual machine, although I will have to learn it sometime because I am interested in possibly resurrecting PrettyGCode, and have been having a look at the philosophy and structure in order to decide whether it’s doable.

It’s the early hours of the morning your end, so I propose to start now in any case. If there is anything that I should do, or you want to know about, during the print, please let me know

Cheers
Nic

nicclift, 2025-09-19:
Update: - Interesting

Re-sliced, exported GCODE to OctoPrint. Mac and File Manger agree 234.54MB and 234.6MB respectively
Tried visualising the gcode - stopped as soon as “Analysis” reached and did not proceed
Ran Cleanup Plugin Data on Plugin Manager - calls for restart
After restarting was surprised to see this screen, never previously encountered:

￼
Accepting the warning (spurious, the file prints fine) and proceeding, the “Analysing” step now freezes at 51%

￼
I click on the spanner for “Settings”, and immediately this image appears of the dragon. But the “Alnlyzing” has not progressed further. 
￼
More of a worry, though, is this:
The file size is still reported correctly, but the print time is indicated to be 16h. No way!

￼
PrusaSlicer estimates roughly 1.5 days - approx double this amount of time (if you are unfamiliar, bottom right of the scr:

￼
I deleted the file from OctoPrint (now no files in memory), disabled the gcode visualiser plugin, reset OctoPrint, and sent the file from PrusaSlicer again. All sizes still the same.
When I loaded the file and started to print, no estimate of print time. However, I had set the deck plate slightly askew, and when I stopped the print to correct this, the following updated information became available:
Once again, an estimate of 16h of print time.

￼
I have brought a copy of the GCODE back from OctoPrint, and have dropped it on the latest link you gave me.

In case this is an issue with running OctoPrint on a Pi, here is the info in case that helps. It’s a modestly spec’d Pi5, which I assume is more than adequate for this job?:

Category

Detail

Architecture

aarch64 (ARM 64-bit)

CPU Model

ARM Cortex-A76 (r4p1)

Cores

4 cores (1 thread per core, single cluster)

Clock Speed

Min: 1.5 GHz · Max: 2.4 GHz (currently scaling ~67%)

BogoMIPS

108.00

CPU Flags

fp, asimd, evtstrm, aes, pmull, sha1, sha2, crc32, atomics, asimdhp, cpuid, asimdrdm, lrcpc, dcpop, asimddp

Caches

L1d: 256 KiB (×4) · L1i: 256 KiB (×4) · L2: 2 MiB (×4) · L3: 2 MiB (shared)

NUMA Report

Reports 8 NUMA nodes (0–7) but all mapped to CPUs 0-3 (artifact of kernel reporting, not true NUMA)

Vulnerabilities

All major ones (Spectre, Meltdown, etc.) reported as Not affected or Mitigated

Memory (GPU split)

GPU: 4 MB · ARM: 1020 MB (≈1 GB accessible to system)

Storage (/)

117 GB total · 8.4 GB used · 103 GB free (~8% usage)

Cheers
Nic

jneilliii, 2025-09-19:
Unfortunately, embedded images in email replies don't transfer to GitHub.

nicclift, 2025-09-19:
> "I'm confused by its file size, which doesn't match the 248MB mentioned before, but rather the 235MB I already saw yesterday on the file from the slicer (from my understanding) and apparently also the new file you now sliced freshly."

That file size was the result of a bgcode file being sent to OctoPrint, which converted it to the larger size. I have not done any test to confirm whether or not this file would do the same - as per your request I sent the file this time as GCODE.

I have tried to print this dragon on a number of occasions over the last several weeks. Sometimes I re-sliced. Before reporting this issue I had deleted a whole load of failed attempts, and this morning deleted the rest, believing that the two versions I have sent you were sufficient (the first as the originally sliced bgcode and also converted back to GCODE using PrusaSlicer; the second re-sliced as GCODE only). The dragon is saved as a Project - but this morning's file has different temperatures for some of the filament, which previously I was managing in the dashboard. This may explain different file sizes. 

The backlog of failed attempts has been cleared. Sorry if this poses a problem!

When I saw the 16h estimate this morning (as opposed to PrusaSlicer and personal experience both indicating around twice as long), it looked all too similar to previous failures. As a result, I cancelled and re-launched from a USB drive and did not use OctoPrint. I had not realised that the time estimate in OctoPrint could be so different, otherwise I would have persisted. 

If the problem is one particular version of my sliced code, then there is unlikely to be an issue with OctoPrint, other than bad luck with my faulty sliced code triggering a stop rather than an alarm. Odd that the reported % completion could be so different - roughly a factor of 2 - between reported on OctoPrint and both visible and reported on the printer itself, though.

If the problems of premature completion with no messages, and reporting of progress that is approx double real, are something to do with size/complexity/other characteristic of otherwise OK sliced code, then presumably your modelled run/virtual machine will confirm or contradict this - even if the sliced file is not exactly the same as the one that prompted my reporting this issue. I will do another run using OctoPrint when this current run finishes and let you know the outcome.

foosel, 2025-09-25:
A defective control board certainly sounds like it could also have been a factor here.

In any case, let's close this for now, and if you should see it again, just ping me here and we'll reopen both the issue and the investigation :)